goodix-fp-linux-dev/goodix-fp-dump #2: 5385 usb timeout

Roy-Ermers, 2021-08-20:
```
Type 9243 to continue and confirm that you are not a bot: 9243
__init__(21381, <class 'protocol.USBProtocol'>, 5)
Found Goodix device: "Goodix Fingerprint Device " from "HTMicroelectronics" on bus 1 address 6.
Found interface data: 1
Found endpoint in: 0x81
Found endpoint out: 0x3
empty_buffer()
nop()
firmware_version()
Traceback (most recent call last):
  File "/home/roy/source/goodix-fp-dump/./run_5385_nonsafe.py", line 3, in <module>
    main(0x5385)
  File "/home/roy/source/goodix-fp-dump/driver_538.py", line 285, in main
    firmware = device.firmware_version()
  File "/home/roy/source/goodix-fp-dump/goodix.py", line 552, in firmware_version
    check_message_protocol(check_message_pack(self.protocol.read()),
  File "/home/roy/source/goodix-fp-dump/protocol.py", line 126, in read
    return self.device.read(self.endpoint_in, size, timeout).tobytes()
  File "/usr/local/lib/python3.9/site-packages/usb/core.py", line 1029, in read
    ret = fn(
  File "/usr/local/lib/python3.9/site-packages/usb/backend/libusb1.py", line 846, in bulk_read
    return self.__read(self.lib.libusb_bulk_transfer,
  File "/usr/local/lib/python3.9/site-packages/usb/backend/libusb1.py", line 954, in __read
    _check(retval)
  File "/usr/local/lib/python3.9/site-packages/usb/backend/libusb1.py", line 602, in _check
    raise USBTimeoutError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBTimeoutError: [Errno 110] Operation timed out
```

mpi3d, 2021-08-20:
Well, this means that 538 driver doesn't work on this device.

Sorry.

mpi3d, 2021-08-20:
Removed device: 3ff82d99301b0de0af9391afda2c190f15c15736

closed 2021-08-20 by mpi3d