goodix-fp-linux-dev/goodix-fp-dump #5: How dangerous is it really?

atar-axis, 2021-11-03:
There is a big warning notice prior to using this dump tool,
it says that you could possibly break your fingerprint reader by using it.
How dangerous is it really and is there any way to flash back the original fw,
e.g. by using Windows?

mpi3d, 2021-11-04:
The process should be safe. We took time to reverse engineer this specific part of the Windows driver. So this program should have the same behavior as the Windows one. It can even be safer to use this one than the one from Goodix (Their way of programming is very weird). So you shouldn't be worried if you run this program without any modifications.

Also, the device use two firmware at different level. We are only flashing the one that permit to get images. The other is specific for the flashing routine of the first one. So in case that the first firmware flash fail there is still the second firmware that will work, and so we will be able to still communicate with the device. In any case, Windows driver will be able to flash the original firmware as long as the device respond.

But we are still flashing a firmware, so there is always a risk that something goes wrong and brick the device. So we prefer to warn the user about this flashing process.

By the way, we are flashing an unmodified firmware, so we are flashing the original firmware from Goodix as the Windows driver does.

I hope that was clear.

closed 2021-11-04 by mpi3d

atar-axis, 2021-11-04:
Very clear, thanks a lot! Is there anything where I can be of any help? I am not new to driver programming but I am far from being an expert, too. Anyway please let me know if and how I can help :)

mpi3d, 2021-11-04:
See you on the Discord server.