goodix-fp-linux-dev/goodix-fp-dump #11: ModuleNotFoundError: No module named 'spidev'

khandakershahi, 2022-02-18:
Hi,
I have Dell inspiron 5505.
When I try to run your code it give me error with `ModuleNotFoundError: No module named 'spidev'`

 ```
sudo python3 run_538d.py                                          
Traceback (most recent call last):
  File "/home/shahi/goodix-fp-dump/run_538d.py", line 1, in <module>
    from driver_53xd import main
  File "/home/shahi/goodix-fp-dump/driver_53xd.py", line 9, in <module>
    from goodix import FLAGS_TRANSPORT_LAYER_SECURITY_DATA, Device
  File "/home/shahi/goodix-fp-dump/goodix.py", line 13, in <module>
    from protocol import Protocol, USBProtocol
  File "/home/shahi/goodix-fp-dump/protocol.py", line 13, in <module>
    from spidev import SpiDev
ModuleNotFoundError: No module named 'spidev'
```

mpi3d, 2022-02-20:
Well for now the driver is just able to read an image from the sensor.
But there is still much work to do to be able to enroll fingerprints.

newtonsart, 2022-03-02:
just do
```
sudo pip3 install spidev
```
that worked for me

steqtk, 2022-03-07:
I have this after installing `sudo pip3 install spidev`:
```
Traceback (most recent call last):
  File "run_5110.py", line 1, in <module>
    from driver_51x0 import main
  File "/home/steq/Загрузки/goodix-fp-dump/driver_51x0.py", line 9, in <module>
    from goodix import FLAGS_TRANSPORT_LAYER_SECURITY, Device
  File "/home/steq/Загрузки/goodix-fp-dump/goodix.py", line 13, in <module>
    from protocol import Protocol, USBProtocol
  File "/home/steq/Загрузки/goodix-fp-dump/protocol.py", line 14, in <module>
    from periphery import CdevGPIO
ModuleNotFoundError: No module named 'periphery'
```
```
$ sudo pip3 install periphery
ERROR: Could not find a version that satisfies the requirement periphery (from versions: none)
ERROR: No matching distribution found for periphery
```

newtonsart, 2022-03-07:
> I have this after installing `sudo pip3 install spidev`:
>
> ```
> Traceback (most recent call last):
>   File "run_5110.py", line 1, in <module>
>     from driver_51x0 import main
>   File [...]

if you are using archlinux: ```sudo pacman -S python-periphery```
if you are using other distribution search how to install python-periphery on *your distribution*

mpi3d, 2022-03-15:
Closing because of inactivity

closed 2022-03-15 by mpi3d