goodix-fp-linux-dev/goodix-fp-dump #12: "Errno Device not found"

E34rrsff, 2022-02-18:
I'm not sure how to fix this error

`TimeoutError: [Errno Device not found] -5: 19`

mpi3d, 2022-02-20:
Could you give more info about the sensor. Also can you provide the full output of the program?

E34rrsff, 2022-02-21:
I'm pretty sure it's the 55b4, but it doesn't get listed in lsusb or lspci. I have the Lenovo Yoga C940 and I flashed some firmware to get the speakers working.

Here's the output:

#######################################################
This program might break your device.
Consider that it may flash the device firmware.
Continue at your own risk.
But don't hold us responsible if your device is broken!
Don't run this program as part of a regular process.
#######################################################
Type 5724 to continue and confirm that you are not a bot: 5724
__init__(21940, <class 'protocol.USBProtocol'>, 5)
Traceback (most recent call last):
  File "/home/e34/Git Clones/goodix-fp-dump/run_55b4.py", line 3, in <module>
    main(0x55b4)
  File "/home/e34/Git Clones/goodix-fp-dump/driver_55x4.py", line 218, in main
    device = init_device(product)
  File "/home/e34/Git Clones/goodix-fp-dump/driver_55x4.py", line 45, in init_device
    device = Device(product, USBProtocol)
  File "/home/e34/Git Clones/goodix-fp-dump/goodix.py", line 167, in __init__
    self.protocol: Protocol = protocol(0x27c6, product, timeout)
  File "/home/e34/Git Clones/goodix-fp-dump/protocol.py", line 65, in __init__
    raise TimeoutError("Device not found", -5, 19)
TimeoutError: [Errno Device not found] -5: 19

mpi3d, 2022-02-23:
If the device doesn't show in your `lsusb`, this mean that the device is not present on your laptop.
Also, if your device doesn't show in your `lsusb`, this script will not work anyway. This is why the program raise `TimeoutError: [Errno Device not found] -5: 19`

So maybe your device is an SPI one.
Could you provide the result of `/sys/bus/spi/devices`?

E34rrsff, 2022-02-24:
There's nothing in the directory

mpi3d, 2022-02-24:
Your device doesn't seems to be a Goodix one then.
This repo is only for Goodix devices.
So I can't do much more then.

closed 2022-02-24 by mpi3d

E34rrsff, 2022-03-10:
I've confirmed that it is definitely is a Goodix sensor. On the support page for my laptop, it has a Windows driver with the names "Goodix" and "Synaptics". It also doesn't get detected in Windows for some reason, even though it worked before...

mpi3d, 2022-03-10:
Well I can't do anything as long as the device doesn't show up in your `lsusb`.

E34rrsff, 2022-03-12:
So, I guess it was something with the newly flashed BIOS. There's just lots of options in it, and none of them say anything about a "fingerprint sensor," just some device IDs and some stuff I don't really know about, but it works now. Thanks for the help.

useHTML5, 2023-03-03:
I have a same error.

```
workeuser@pc:~$ lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 003: ID 5986:2146 Acer, Inc Integrated RGB Camera
Bus 003 Device 002: ID 27c6:550a Shenzhen Goodix Technology Co.,Ltd. FingerPrint
Bus 003 Device 004: ID 8087:0026 Intel Corp. AX201 Bluetooth
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

mpi3d, 2023-03-03:
> I have a same error.
> 

Your device is not supported.

ZhaoCake, 2024-05-10:
> > I have a same error.
> 
> Your device is not supported yet.

excuse me, am I the same? One year passed , how about now?

[image]

[image]