goodix-fp-linux-dev/goodix-fp-dump #14: 55a4 timeout

sweetquiet, 2022-03-14:
Type 2576 to continue and confirm that you are not a bot: 2576
__init__(21924, <class 'protocol.USBProtocol'>, 5)
Found Goodix device: "Goodix FingerPrint Device" from "Generic" on bus 3 address 3.
Found interface data: 0
Found endpoint in: 0x82
Found endpoint out: 0x1
empty_buffer()
nop()
firmware_version()
Traceback (most recent call last):
  File "run_55a4.py", line 3, in <module>
    main(0x55a4)
  File "/home/wangning/goodix-fp-dump/driver_55x4.py", line 221, in main
    firmware = device.firmware_version()
  File "/home/wangning/goodix-fp-dump/goodix.py", line 573, in firmware_version
    check_message_protocol(check_message_pack(self.protocol.read()),
  File "/home/wangning/goodix-fp-dump/protocol.py", line 130, in read
    return self.device.read(self.endpoint_in, size, timeout).tobytes()
  File "/usr/local/lib/python3.8/dist-packages/usb/core.py", line 1029, in read
    ret = fn(
  File "/usr/local/lib/python3.8/dist-packages/usb/backend/libusb1.py", line 846, in bulk_read
    return self.__read(self.lib.libusb_bulk_transfer,
  File "/usr/local/lib/python3.8/dist-packages/usb/backend/libusb1.py", line 954, in __read
    _check(retval)
  File "/usr/local/lib/python3.8/dist-packages/usb/backend/libusb1.py", line 602, in _check
    raise USBTimeoutError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBTimeoutError: [Errno 110] Operation timed out

mpi3d, 2022-03-14:
#13 might help you solve this

sweetquiet, 2022-03-15:
This not worked. Thanks for your time!

closed 2022-03-15 by mpi3d