goodix-fp-linux-dev/goodix-fp-dump #15: [goodix.py decoding error] UnicodeDecodeError: 'utf-8' codec can't decode byte 0xee in position 22

vicw0ng-hk, 2022-03-18:
Running `$ sudo python goodix-fp-dump/run_55b4.py`

Getting the results:

```bash
#######################################################
This program might break your device.
Consider that it may flash the device firmware.
Continue at your own risk.
But don't hold us responsible if your device is broken!
Don't run this program as part of a regular process.
#######################################################
Type 3294 to continue and confirm that you are not a bot: 3294
__init__(21940, <class 'protocol.USBProtocol'>, 5)
Found Goodix device: "Goodix FingerPrint Device" from "Generic" on bus 3 address 4.
Found interface data: 0
Found endpoint in: 0x82
Found endpoint out: 0x1
empty_buffer()
nop()
firmware_version()
Firmware: GF3268_RTSEC_APP_10024
preset_psk_read(3137470471, None, None)
Valid PSK: False
get_iap_version(25)
Traceback (most recent call last):
  File "/home/victor/finger/goodix-fp-dump/run_55b4.py", line 3, in <module>
    main(0x55b4)
  File "/home/victor/finger/goodix-fp-dump/driver_55x4.py", line 230, in main
    iap = device.get_iap_version(25)
  File "/home/victor/finger/goodix-fp-dump/goodix.py", line 879, in get_iap_version
    return message.split(b"\x00")[0].decode()
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xee in position 22: invalid continuation byte
```

Python version: `3.10.2`

Related modules:
```
crcmod==1.7
python-periphery==2.3.0
pyusb==1.2.1
spidev==3.5
```

:heart: Thank you for this repo :heart:

vicw0ng-hk, 2022-03-18:
After chaging line `879` of `oodix.py` to 
```python3
return message.split(b"\x00")[0].decode(encoding="ISO-8859-1")
```

, it gets IAP `MILAN_RTSEC_IAP_10013îVW`. Even by commenting out lines `233` - `237` of `driver_55x4.py` to force it to run, I get time out
```
#######################################################
This program might break your device.
Consider that it may flash the device firmware.
Continue at your own risk.
But don't hold us responsible if your device is broken!
Don't run this program as part of a regular process.
#######################################################
Type 8987 to continue and confirm that you are not a bot: 8987
__init__(21940, <class 'protocol.USBProtocol'>, 5)
Found Goodix device: "Goodix FingerPrint Device" from "Generic" on bus 3 address 5.
Found interface data: 0
Found endpoint in: 0x82
Found endpoint out: 0x1
empty_buffer()
nop()
firmware_version()
Firmware: MILAN_RTSEC_IAP_10013
preset_psk_read(3137470471, None, None)
Valid PSK: True
get_iap_version(25)
Traceback (most recent call last):
  File "/home/victor/finger/goodix-fp-dump/run_55b4.py", line 3, in <module>
    main(0x55b4)
  File "/home/victor/finger/goodix-fp-dump/driver_55x4.py", line 230, in main
    iap = device.get_iap_version(25)
  File "/home/victor/finger/goodix-fp-dump/goodix.py", line 874, in get_iap_version
    check_message_pack(self.protocol.read()), COMMAND_GET_IAP_VERSION)
  File "/home/victor/finger/goodix-fp-dump/protocol.py", line 130, in read
    return self.device.read(self.endpoint_in, size, timeout).tobytes()
  File "/home/victor/finger/venv/lib/python3.10/site-packages/usb/core.py", line 1029, in read
    ret = fn(
  File "/home/victor/finger/venv/lib/python3.10/site-packages/usb/backend/libusb1.py", line 846, in bulk_read
    return self.__read(self.lib.libusb_bulk_transfer,
  File "/home/victor/finger/venv/lib/python3.10/site-packages/usb/backend/libusb1.py", line 954, in __read
    _check(retval)
  File "/home/victor/finger/venv/lib/python3.10/site-packages/usb/backend/libusb1.py", line 602, in _check
    raise USBTimeoutError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBTimeoutError: [Errno 110] Operation timed out
```

Guess it doesn't work after all. Should be similar to #14. Thanks anyway.

mpi3d, 2022-03-19:
Did you tried to do this: https://github.com/mpi3d/goodix-fp-dump/issues/13#issuecomment-1057802514

vicw0ng-hk, 2022-03-19:
Still timed out. Thanks though.

mpi3d, 2022-03-19:
Ok. So I have no clue of what that could be.