goodix-fp-linux-dev/goodix-fp-dump #16: ValueError: Invalid OTP

fyangami, 2022-03-20:
__init__(21021, <class 'protocol.USBProtocol'>, 5)
Found Goodix device: "FingerPrint" from "Goodix" on bus 3 address 4.
Found interface data: 0
Found endpoint in: 0x83
Found endpoint out: 0x1
empty_buffer()
nop()
firmware_version()
Firmware: GFUSB_GM168SEC_APP_10019
preset_psk_read(3137470465, 32, 0)
Valid PSK: True
reset(True, False, 20)
read_sensor_register(0, 4)
read_otp()
Traceback (most recent call last):
  File "/home/fyang/Downloads/goodix-fp-dump/run_521d.py", line 3, in <module>
    main(0x521d)
  File "/home/fyang/Downloads/goodix-fp-dump/driver_52xd.py", line 344, in main
    run_driver(device)
  File "/home/fyang/Downloads/goodix-fp-dump/driver_52xd.py", line 133, in run_driver
    raise ValueError("Invalid OTP")

mpi3d, 2022-03-20:
As you might have noticed, the driver is still in developement. So many feature are still missing and doesn't works. You need to stay patient.

fyangami, 2022-03-20:
I replaced the value of `OPT` with a constant value and ran the script, which seemed to work normally😃

mpi3d, 2022-03-20:
Ok. Thanks

closed 2022-03-20 by mpi3d

sashkent3, 2022-04-10:
@fyangami Could you please elaborate what constant value did you replace the OTP with? Exact code snippet would be great to help me and others.

fyangami, 2022-04-10:
> @fyangami Could you please elaborate what constant value did you replace the OTP with? Exact code snippet would be great to help me and others.
just is in source file `driver_52xd.py` like this:
```python
        # otp = device.read_otp()
        # replace with 
        otp = 'ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff'
        assert otp >= 64
        if len(otp) < 64:
            raise ValueError("Invalid OTP")

        # OTP 1: 4e4c4d31372e0000b9828da2a2d73e09
        #        08196896800000ee6014a774a060b614
        #        ea2704009b0056f007212723a1a7a300
        #        00000000000000000000000083760000
        # OTP 2: 4e4b35594c2e00002983759520190009
        #        08274c96800000f0103cae6ea010593c
        #        ea2f04009c0053f00729312ba8b0aa00
        #        000000000000000000000000f3830000
```
This may lead to other problems.