goodix-fp-linux-dev/goodix-fp-dump #20: Driver for 27c6:5125?

Kajumov, 2022-05-16:


mpi3d, 2022-05-16:
This is device is unknown. Can you share some info about it on the discord server?
https://discord.com/invite/6xZ6k34Vqg

closed 2022-05-19 by mpi3d

hsxsix, 2022-05-25:
This device is used in matebook x pro 2021, my computer is it. How do I share this device information with you?

mpi3d, 2022-05-25:
Join us on the discord server and post your `lsusb` in #5125 channel to start.

woodvale, 2022-12-03:
Hi, I stumbled upon this issue while searching if I can get my fp working. Sorry, don't use Discord, was there any progress on this? 

`lsusb`:
```
Bus 003 Device 003: ID 27c6:5125 Shenzhen Goodix Technology Co.,Ltd. Goodix Fingerprint Device 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            2 Communications
  bDeviceSubClass         1 Direct Line
  bDeviceProtocol         1 
  bMaxPacketSize0        64
  idVendor           0x27c6 Shenzhen Goodix Technology Co.,Ltd.
  idProduct          0x5125 
  bcdDevice            2.00
  iManufacturer           1 Shenzhen Goodix Technology Co.,Ltd.
  iProduct                2 Goodix Fingerprint Device 
  iSerial                 3 00000000001A
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0043
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x60
      (Missing must-be-set bit!)
      Self Powered
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         2 Communications
      bInterfaceSubClass      1 Direct Line
      bInterfaceProtocol      1 
      iInterface              0 
      CDC Header:
        bcdCDC               1.10
      CDC Call Management:
        bmCapabilities       0x00
        bDataInterface          1
      CDC ACM:
        bmCapabilities       0x02
          line coding and serial state
      CDC Union:
        bMasterInterface        0
        bSlaveInterface         1 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              16
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass        10 CDC Data
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
Device Status:     0x0001
  Self Powered
```

mpi3d, 2022-12-03:
No progress on it yet.

kawaegle, 2023-01-27:
is there any way to help about ? 

like a start-guide to reverse engineering it or just a template of how the other are done ? 

I really want to help but IDK where start my investigation on it

jenujari, 2026-08-18:
## 27c6:5125 — protocol confirmed, firmware extracted, white-box PSK generator working (and a warning: I bricked my sensor)

Posting everything I found on the 5125 (HONOR MagicBook NMH-WCX9). Most of the pieces needed for a driver are now solved. **I also destroyed my own sensor doing it — the last section explains exactly how, so nobody repeats it.** Please read that part first if you have a working 5125.

### The device

- `27c6:5125`, CDC-ACM, bulk `0x01` OUT / `0x81` IN, 64-byte packets — same transport layout as the 51x7.
- **It speaks the standard Goodix message framing.** Every command in the known set ACKs and replies correctly.
- Sensor register `0x0000` = `a2 04 25 00` — an **exact match** with the 5117/5110, which `driver_51x7.py` already checks for. Same 80x88 sensor silicon.
- Firmware as shipped: `GF_ST411SEC_APP_12508`. Bootloader: `MILAN_ST411SEC_IAP_12501`.
- Firmware readback (`0xf2`) is ACKed but returns no data in application mode, so the running image cannot be backed up.
- `get_iap_version` (`0xf6`) returns a **single status byte** in application mode, not a version string. The bootloader only identifies itself *after* the app is erased — so you cannot check what bootloader you have before committing to the erase.

### Vendor driver — obtainable without Windows

Microsoft Update Catalog serves the OEM package as a plain `.cab`, extractable with `bsdtar` on Linux:

- Search the catalog for the hardware id `USB\VID_27C6&PID_5125`.
- The match is **"Goodix FP - Biometric - 1.1.125.20"**. Its `gfusb.inf` explicitly lists `USB\VID_27C6&PID_5125`.
- `gfusb.dll` (5.5 MB) targets `GF_ST411SEC_APP_12512`, embeds mbedTLS (`ssl_srv.c`), and build paths read `huawei_win\winfpcode\Milan_Watt\MilanSpi`.

### Firmware images are embedded in gfusb.dll

Each image sits behind a length-prefixed header: `0x15` + 21-char version string + image bytes. The next header bounds the previous image.

`GF_ST411SEC_APP_12512` — **127482 bytes, load base `0x08020000`**, appearing 19 times, all byte-identical. Validation: all 49 vector-table handlers resolve inside the image and are Thumb-aligned; a wrong base or size scatters them outside. `SP=0x20020000`, reset `0x0803c135`, CRC-32/MPEG-2 `57f6beb8`.

(Note the sibling `GF_HC460SEC_APP_12509` images are *not* cleanly bounded this way — they bleed into DLL string data, so the gap-to-next-header is only an upper bound in general. It happens to be exact for the ST411 image because an HC460 header always follows it.)

### The PSK, and how the key hash works

`preset_psk_read(0xbb020003)` returns a hash, and the relation is:

```
reported_hash = SHA256(white_box_blob)
```

Verified against the public 51x7 constants: `SHA256(PSK_WHITE_BOX) == PMK_HASH`. **This means candidate blobs can be checked offline against a device's reported hash without writing anything.**

The 5125 ships with a **per-unit** factory blob (mine hashed `797ce2b7...`). It is not in any driver binary, and only the hash slot is readable — every other slot selector is rejected.

### The vendor driver generates the white box; it does not look up a key

`production_write_key` in `gf_production.c` (function at RVA `0x412a8`, PSK write at `0x416e4`) logs its own steps:

```
0. generate random psk
1. seal psk               (gf_seal_data — DPAPI, host-side storage only)
2. process encrypted psk  (p_tlv_psk_wb)
3. write to mcu           (preset_psk_write, flags 0xbb010003)
```

Feasibility notes: **no SGX anywhere** (zero `ENCLU` opcodes, no `sgx_`/enclave strings — `gf_sgx_seal_data` is a legacy name), and the only crypto imports are `CryptGenRandom` and DPAPI `CryptProtectData`/`CryptUnprotectData`. DPAPI stores the plaintext PSK in a host file (`production_get_host_psk_data`, `fail to unseal file data`); it is not involved in the MCU blob.

So the wrap is self-contained computation inside `gfusb.dll`.

### Running the white-box wrap on Linux

The wrap is `sub_180006c20(psk, 0x20, out, &out_len)` with `out_len` starting at `0x7f8`. I ran it under Unicorn: map `gfusb.dll` at its preferred base `0x180000000` (so no relocations), replace Windows imports with sentinels backed by a bump heap, and call it directly.

Results:

- Returns 0, emits a **102-byte** blob (`0x7f8` is only buffer capacity, not output size)
- Touches only `HeapAlloc`/`HeapFree`
- **Deterministic**, and varies with the PSK
- **Takes no device input — the blob is a pure function of the PSK, with no per-unit binding**

Structure looks like `32 bytes | 02 ff 20 00 00 00 | 32 | 32`. It does not reproduce the 96-byte public 51x7 blob, which is expected — that is the older format.

The TLV written to the MCU is exactly `[type=0xbb010003][length][blob]`, which matches `preset_psk_write(flags, payload)`.

**This is the piece that should make a 5125 driver possible**: generate a PSK, wrap it, provision it, then do TLS-PSK with the PSK you chose. The device negotiates `0x00A8` (`TLS_PSK_WITH_AES_128_GCM_SHA256`).

### How I bricked mine — please do not repeat this

The order matters enormously, and I got it wrong.

1. While the sensor was **healthy**, I wrote the public 51x7 96-byte white box over the per-unit factory blob. It was accepted in application mode (`status 0x00`) and the reported hash changed as expected. **This destroyed the only key the sensor had** — the factory blob was never readable, so there was nothing to restore.
[... cut ...]
Everything above is reproducible from the public `.cab` plus `goodix-fp-dump`. Happy to share the extraction and emulation scripts (firmware carver, Unicorn harness for the wrap routine, flasher with preflight) if they are useful — say the word and I will clean them up and put them somewhere.

I no longer have working hardware to test against, so I cannot take this further myself. Hopefully the protocol confirmation, the extracted `12512` image, and especially the white-box generator save someone else the effort.

hsxsix, 2026-08-19:
@jenujari This is an incredible writeup — thank you for documenting the bricking path so clearly. I have a **working `27c6:5125`** on a Huawei MateBook (`MACHD-WXX9`, same CDC-ACM / bulk `0x01` OUT / `0x81` IN 64-byte layout you described), and I'd like to help push this forward while there's still healthy hardware to test against.

Two asks:

1. **Please do share the scripts** — the firmware carver, the Unicorn harness for the white-box wrap routine, and the flasher-with-preflight. A gist or a branch is totally fine. I'll run them against my unit and report back.

2. Before I touch anything, I want to make sure I only do **read-only / non-destructive** operations and never overwrite the per-unit factory blob. Could you confirm the safe-dump plan? My understanding from your post is that these are all read-only:
   - `preset_psk_read(0xbb020003)` to get my per-unit reported hash
   - reading sensor register `0x0000` (expecting `a2 04 25 00`)
   - `get_iap_version` (`0xf6`) / firmware version string in application mode
   - `query_mcu_state` (`0xAE`)

   And the hard rule I'm taking from your experience: **do not `preset_psk_write` and do not erase/flash** while on stock `12508`, since the factory blob is unrecoverable once overwritten. Is that the right line to draw?

My plan is exactly the order you recommended: build and validate the white-box generator first, confirm offline that `SHA256(my_blob)` matches my device's reported hash, and only then even consider provisioning — ideally staying on stock `12508` the whole time since it accepts PSK writes in application mode without an erase.

Happy to be the test rig for a driver. What device-side data would be most useful for you if I dump it (safely)?

Thomas97460, 2026-08-20:
@hsxsix @jenujari Following up on the safe-dump plan proposed above — I have a working `27c6:5125` too, on a Huawei MateBook 16. Ran the read-only checks (chip_id, firmware version, PSK hash, MCU state — no PSK write, no erase, no flash):

```
chip_id (0x0000, 4 bytes)         = a2042500   -> matches 51x0/51x7 family
firmware_version                  = GF_ST411SEC_APP_12508
preset_psk_read(0xbb020003) hash  = 81f7283a6f113a1dcef97f35949d0dbf271628f852efc84e3053140414eb907e
query_mcu_state(0x55)             = 01023500003200002025130011b00000
```

Stock firmware, healthy per-unit PSK hash (different from jenujari's, as expected for a per-unit blob).

Happy to be a second test rig on healthy stock 12508 hardware. I adapted `driver_51x7.py`'s non-destructive bring-up sequence (nop/enable_chip/reset) plus only the read-side commands (`read_sensor_register`, `firmware_version`, `preset_psk_read`, `query_mcu_state`) into a small read-only script to stay strictly non-destructive.

Also looking into extracting/emulating the PSK white-box wrap routine from the OEM `gfusb.dll` (matching jenujari's approach) to get an independent reproduction before anyone else risks a write. Will report back if that pans out.

jenujari, 2026-08-21:
@hsxsix @Thomas97460 — scripts are up: https://gist.github.com/jenujari/5798494ff5ae7bc7be22a1f7ffcbfd78

Contents: `goodix_proto.py` (framing), `probe.py` (read-only probe), `extract_firmware.py` (carves the embedded images out of `gfusb.dll`), `whitebox.py` (Unicorn harness for the wrap routine), `flash.py` (**destructive**, preflight + gated behind an explicit flag).

### Your read-only list — confirmed

All four are read-only, yes:

- `preset_psk_read(0xbb020003)` — hash only; every other slot selector is rejected
- `read_sensor_register(0x0000, 4)`
- `firmware_version` (`0xa8`) / `get_iap_version` (`0xf6`)
- `query_mcu_state` (`0xAE`)

Also read-only, worth grabbing: `read_otp` (`0xa6`). Note `get_iap_version` returns a **single status byte** in application mode, not a string — harmless, just useless until you're in IAP.

`nop` / `enable_chip` / `reset` are runtime-only and non-persistent, so the `driver_51x7` bring-up prelude is fine.

### One correction to my own advice

I said "provision on stock 12508 since it accepts PSK writes without an erase". **Do not treat that as safe yet.** Two reasons:

1. The blob `whitebox.py` produces comes from the `1.1.125.20` driver, whose firmware target is `12512`, **not `12508`**. I never tested it against `12508`. It may simply not be valid for your running firmware.
2. My device died specifically because I wrote a key valid for *one* firmware and then ran *different* firmware. Writing a mismatched blob and staying on `12508` is less bad than what I did, but it is still untested and still destroys your factory blob.

Refinement worth knowing though: the factory blob is **not** sacred. `production_write_key` shows Windows itself generates a fresh random PSK and re-provisions — so the per-unit blob is something the vendor driver replaces, not a one-time secret. What actually bricked mine was the firmware/key mismatch afterwards, not the overwrite by itself.

### Most useful thing you could do (still zero writes)

If either of you has Windows on the machine: capture the USB traffic while the Goodix driver enrolls/initialises, and pull out the `preset_psk_write` frame (`flags = 0xbb010003`). That gives us **the exact blob the vendor driver writes to a real 5125**. Compare its length and structure against `whitebox.py` output — if they match in shape, the emulator is validated with nothing at risk. That single capture is worth more than any amount of further static analysis.

Failing that, the useful safe dumps are: full `read_otp`, `query_mcu_state`, and your PSK hash (thanks @Thomas97460 — your `01 02 35 00...` on healthy stock is a good baseline; mine reads `01 09 33 00...` bricked, so that byte does look like a real state indicator).

### Suggested order, if you do decide to write

1. Run `whitebox.py <psk-hex>` and note `SHA256(blob)`.
2. Provision it, then immediately `preset_psk_read(0xbb020003)` and confirm the device reports exactly that hash. If it does, the blob was stored verbatim.
3. **Do not flash anything.** Stay on `12508` and go straight to a TLS-PSK handshake with the PSK you chose (device negotiates `0x00A8`). If the handshake completes, you have a working sensor and no firmware was ever touched.

Step 3 is where I went wrong — I reached for the flasher instead. If the handshake fails on `12508`, stop there and report; that tells us the blob format is firmware-specific and the answer is a `12512`-derived path, not a `12109` one.

Happy to help debug from the sidelines — I have no working hardware left.

Thomas97460, 2026-08-28:
Update: I now have an experimental libfprint driver working for `27c6:5125`:

https://gitlab.freedesktop.org/libfprint/libfprint/-/merge_requests/648

Enrollment, positive/negative verification and KDE Plasma lock-screen unlock
were tested on one device. No firmware flashing, `goodix.dat` or manual
calibration file is needed.

Build/install without Nix (after installing the usual libfprint build
dependencies plus OpenSSL headers):

```sh
git clone --branch goodix-5125-oem-image --single-branch \
  https://gitlab.freedesktop.org/Thomas97460/libfprint.git
cd libfprint
meson setup build \
  -Ddrivers=goodix5125 \
  -Ddoc=false \
  -Dgtk-examples=false \
  -Dintrospection=false
meson compile -C build
sudo meson install -C build
sudo ldconfig
sudo systemctl restart fprintd
```

Then enable fprintd/PAM using your distribution and enroll from the desktop
fingerprint settings.

A blank sensor is paired automatically. If it is already paired by Windows,
the matching PSK must be preserved; the driver deliberately refuses to
overwrite a foreign pairing. Do not use `GOODIX5125_PROVISION_PSK=random`
unless you accept breaking the Windows pairing.

hsxsix, 2026-08-28:
@Thomas97460 — this is a fantastic milestone, congratulations. A working `27c6:5125` driver with enroll + verify + lock-screen unlock and **no firmware flashing** is exactly the outcome we were hoping for.

I want to make sure I understand the pairing model before I try it, because my unit is a **dual-boot machine whose 5125 is already paired by Windows Hello** (healthy stock `12508`). From my earlier USB capture I established that a normal Windows enrollment never touches the PSK slot, so I have no way to recover the vendor/Windows PSK — the "sniff it out" path is a dead end on an already-provisioned device.

So my situation falls squarely into the case your note warns about:

> If it is already paired by Windows, the matching PSK must be preserved; the driver deliberately refuses to overwrite a foreign pairing.

Could you clarify how that plays out in practice for a Windows-paired unit:

1. **Can the driver actually *use* an existing Windows pairing** (verify/unlock against it) — i.e. does it need me to supply the Windows PSK somehow, or can it operate against a foreign pairing without knowing that key? Or does "preserve the pairing" just mean "it won't clobber it, but it also can't use it," so a Windows-paired sensor effectively **can't** be driven on Linux without re-provisioning?

2. **If re-provisioning is the only way to get Linux working** (a fresh Linux PSK): is that break of the Windows pairing **reversible** — i.e. can Windows Hello simply re-enroll/re-pair afterward and take the sensor back — or does something (the guard, or the OEM image state) make that a one-way trip? I want to be clear on whether it's genuinely "Windows *or* Linux, pick one" vs. "both, but you re-pair whichever side you switch to."

3. Is there **any path to keeping both OSes working** simultaneously (shared/known PSK, importing the Windows pairing, etc.), or is that fundamentally out of reach given the PSK can't be extracted?

I'm deliberately **not** going to run `GOODIX5125_PROVISION_PSK=random` until I understand the above — I'd rather not sacrifice Windows Hello (or risk anything) blindly.

Happy to help test on my end once I know the safe envelope: I have a third confirmed-healthy `12508` unit, Windows on the same machine for before/after comparison, and can capture USB traffic of any specific step if that's useful for validating the flow. Thanks again for pushing this over the line.

Thomas97460, 2026-08-28:
@hsxsix — yes, both Windows and Linux can work with the sensor at the same
time. Linux needs the same 32-byte PSK used by Windows; the sensor hash alone
is only useful for checking that key.

That is the setup I tested. I recovered the PSK offline from
`C:\ProgramData\Goodix\Goodix_Cache.bin` through the Windows machine DPAPI
state, installed it as `/var/lib/fprint/goodix5125/psk`, and the driver
confirmed it against the sensor hash. No PSK write or firmware change was
performed, KDE unlock works, and the Windows pairing remains unchanged.

The fingerprint templates are separate: Linux still does its own enrollment;
only the transport PSK is shared.

So please do not use `GOODIX5125_PROVISION_PSK=random`. That replaces the
single pairing key, and I have not tested Windows recovery after doing so. I
will clean up the read-only DPAPI recovery steps so you can use the same safe
path on your `12508` unit.

hsxsix, 2026-08-29:
@Thomas97460 — thank you for the great work on this. Looking forward to your cleaned-up read-only DPAPI recovery steps; I'll follow them on my `12508` and report back.

H3xano, 2026-09-13:
## 27c6:5125 on HONOR MagicBook 16 (HYM-WXX) — full success with **zero Windows access**, complementing the DPAPI-recovery path above

Adding a data point to this thread: confirms `jenujari`'s protocol/white-box findings and `Thomas97460`'s chip/firmware IDs exactly, and covers the case neither fully solves — **no Windows installation at all** (wiped before ever touching Linux, no dual-boot, no `Goodix_Cache.bin` to recover from).

### Device
- HONOR MagicBook 16, board `HYM-WXX`, SKU `C233`
- `27c6:5125`, chip ID `a2042500` (0x2504) — same ST411SEC silicon as the already-supported 51x7 family
- Firmware `GF_ST411SEC_APP_12508` — matches `Thomas97460`'s and `hsxsix`'s units exactly

### Why "recover the original PSK" isn't an option here
Traced the full pairing flow through the Windows host driver (`gfusb.dll`, from the OEM's official fingerprint driver package). PSK provisioning happens **once per Windows installation**, not once per device at the factory:
- First driver run generates a random 32-byte PSK, white-box-wraps it, writes it to the sensor's MCU, and DPAPI-seals a local cache copy (`Goodix_Cache.bin`)
- DPAPI ties that cache to the *specific Windows user account* that created it
- With that Windows installation gone, there is no `Goodix_Cache.bin`, no DPAPI master key, nothing to recover — this is a genuinely different situation from `hsxsix`'s dual-boot case

So the only path forward was re-provisioning the sensor with a **new** PSK we control, entirely from Linux.

### White-box algorithm — independently cross-validated
Reverse-engineered the wrap routine two ways: emulating the real x86-64 driver code directly (Unicorn, hooking only the CRT allocator functions) and static analysis in Ghidra. Both produced **byte-identical output** to `jenujari`'s independently-derived implementation — same `WB_KDF_KEY`/`WB_GCM_AAD` constants, same HMAC-SHA256 KDF → AES-256-GCM construction, same 102-byte blob layout for a 32-byte PSK. Three independent reverse-engineering efforts landing on identical bytes is about as strong a confirmation as this is going to get.

### The bug that caused rejections (relevant to anyone hitting `preset_psk_write` failures)
The MCU write isn't a single TLV — it's two chained TLV records in one message:
```
[0xBB010002][len=72][64-byte placeholder "seal"][8-byte nonce]
[0xBB010003][len][white-box-wrapped PSK]
```
The first record's length field covers *only* the placeholder+nonce (not the whole message), and the placeholder must be exactly 64 bytes — the MCU firmware doesn't validate its content (it's pure host-side DPAPI-cache bookkeeping the ARM MCU can't process anyway) but it does care about the length/offsets being exactly right. Got this from the community driver's `goodix_psk.c` after two of my own guesses (128-byte placeholder, and a bare single-TLV write) both got rejected with an identical device-side error code — same error regardless of payload shape, which in hindsight was the tell that it was a structural/offset issue, not a policy rejection.

### Result
Wrote a fresh PSK using the corrected format, verified the MCU's returned hash matched the expected `SHA256(white-box-wrap(psk))` exactly, then built and ran [`Rockytkg/goodix-linux-27c6-5125`](https://github.com/Rockytkg/goodix-linux-27c6-5125) against the now-repaired sensor:
- Full TLS-PSK handshake (`TLS-PSK-WITH-AES-128-GCM-SHA256`) completes cleanly
- Real image capture, CRC-validated, clean fingerprint ridge pattern
- `libfprint`/SIGFM integration installed, `fprintd-enroll` / `fprintd-verify` working reliably
- PAM wired up for login + `sudo` (Mint/Cinnamon's `fingwit` profile, not bare `fprintd`)

No firmware flash was needed in this path — same as `Thomas97460`'s MR, which is worth calling out since `jenujari`'s bricking warning is specifically about a firmware-flash gone wrong, not the PSK-write path described here.

### For anyone in the same boat (5125 already paired, no Windows access at all)
1. Extract the OEM's Windows fingerprint driver package (7z can unpack the NSIS installer without running it) to get `gfusb.dll`
2. Either reimplement the white-box wrap from `jenujari`'s gist / `Rockytkg`'s `goodix_psk.c` directly (fastest — the algorithm's already public and cross-confirmed), or re-derive it yourself against your own `gfusb.dll` build as a sanity check
3. Use the corrected TLV format above to write a fresh PSK
4. `Rockytkg/goodix-linux-27c6-5125` handles everything from there — protocol stack, firmware-version check (no update needed if you're already on a released APP firmware), TLS, capture, and full `libfprint` integration

Happy to share the exact Unicorn-emulation harness and provisioning script if useful — didn't want to dump a huge post without knowing if it's wanted given `jenujari`'s gist already covers similar ground.

H3xano, 2026-09-13:
Scripts are up: https://gist.github.com/H3xano/e9488cffc52348dda4265a3b2d053813

`emulate_wb.py` (Unicorn harness running the real driver's wrap routine directly) + `provision_psk.py` (generates a fresh PSK, wraps it, writes the corrected two-record TLV, verifies the MCU hash). Cross-checked against jenujari's implementation — byte-identical output.

arbnorme, 2026-09-25:
Another data point for 27c6:5125, this time with working matching: Huawei MateBook 16 (2021), firmware GF_ST411SEC_APP_12508, Fedora 44.

I wrote (or claude wrote) a libfprint driver (goodixtls5125) that works with the unmodified fprintd: enrol, verify, lock screen, sudo and suspend/resume. The open-source matchers (NBIS, SIGFM) couldn't separate fingers on the 64x80 image, so the driver runs Goodix's own AlgoChicago.dll in-process (loader adapted from goodix-5e0a). The DLL isn't shipped: a script downloads it from the Microsoft Update Catalog and checks its hash. On my device, 15/20 genuine touches were recognised and 0/20 from other fingers were accepted.

Limitation: it uses the all-zero PSK, so it only works on sensors that were never re-provisioned by Windows. No flashing and no PSK writes. Tested on one device only; feedback welcome.

https://github.com/arbnorme/libfprint-goodix5125

arbnorme, 2026-09-25:
I forgot to mention that this repo saved me a lot of tokens: https://github.com/jitendradara12/goodix-5e0a/