goodix-fp-linux-dev/goodix-fp-dump #40: ModuleNotFoundError: No module named 'Crypto'

a2nt, 2024-03-15:
Looks like a python module is missing
```
(.venv) darkmoon :: ~/goodix-fp-dump ‹master› # sudo python3 run_5385.py                                 
Traceback (most recent call last):
  File "/root/goodix-fp-dump/run_5385.py", line 1, in <module>
    import driver_53x5
  File "/root/goodix-fp-dump/driver_53x5.py", line 7, in <module>
    import Crypto.Hash.SHA256
ModuleNotFoundError: No module named 'Crypto'

(.venv) darkmoon :: ~/goodix-fp-dump ‹master› # python --version                                     
Python 3.11.8

```

igorar666, 2024-03-16:
Try 2 install "PyCrypto" or "PyCryptodome"

NameeLesS, 2024-05-27:
You have to use venv python try `sudo .venv/bin/python ./run_5385.py`

pol2095, 2025-01-05:
using `sudo .venv/bin/python ./run_5385.py`, I have this message error :
`AttributeError: module 'Crypto' has no attribute 'Random'`
Thanks

egormanga, 2025-04-04:
- See: https://github.com/goodix-fp-linux-dev/goodix-fp-dump/issues/51#issuecomment-2627524273.

closed 2025-04-04 by egormanga