goodix-fp-linux-dev/goodix-fp-dump #44: ValueError: Invalid OTP

WAHM08, 2024-06-07:
When I execute `sudo python run_521d.py`.  I have the next error:

`
#######################################################
This program might break your device.
Consider that it may flash the device firmware.
Continue at your own risk.
But don't hold us responsible if your device is broken!
Don't run this program as part of a regular process.
#######################################################
Type 1329 to continue and confirm that you are not a bot: 1329
__init__(21021, <class 'protocol.USBProtocol'>, 5)
Found Goodix device: "FingerPrint" from "Goodix" on bus 3 address 4.
Found interface data: 0
Found endpoint in: 0x83
Found endpoint out: 0x1
empty_buffer()
nop()
firmware_version()
Firmware: GFUSB_GM168SEC_APP_10019
preset_psk_read(3137470465, 32, 0)
Valid PSK: True
reset(True, False, 20)
read_sensor_register(0, 4)
read_otp()
Traceback (most recent call last):
  File "/home/wahm/goodix-fp-dump/run_521d.py", line 3, in <module>
    driver_52xd.main(0x521d)
  File "/home/wahm/goodix-fp-dump/driver_52xd.py", line 351, in main
    run_driver(device)
  File "/home/wahm/goodix-fp-dump/driver_52xd.py", line 134, in run_driver
    raise ValueError("Invalid OTP")
ValueError: Invalid OTP

`

I don´t know how to solve. Someone can help me

unjordi, 2024-09-03:
Same error, Same device. here's my output:
`Type 1895 to continue and confirm that you are not a bot: 1895
__init__(21021, <class 'protocol.USBProtocol'>, 5)
Found Goodix device: "FingerPrint" from "Goodix" on bus 3 address 3.
Found interface data: 0
Found endpoint in: 0x83
Found endpoint out: 0x1
empty_buffer()
nop()
firmware_version()
Firmware: GFUSB_GM168SEC_APP_10019
preset_psk_read(3137470465, 32, 0)
Valid PSK: False
mcu_erase_app(50, True)
firmware_version()
Firmware: MILAN_GM168SEC_IAP_10007
preset_psk_read(3137470465, 32, 0)
Valid PSK: False
preset_psk_write(3137404931, b'\xec5\xae:\xbbE\xed?\x12\xc4u\x1f\x1e\\,\xc0[<TR\xe9\x10M\x9f*1\x18dO7\xa0Ko\xd6k\x1d\x97\xcf\x80\xf14_v\xc8O\x03\xff0\xbbQ\xbf0\x8f*\x98u\xc4\x1ee\x92\xcd*/\x9e`\x80\x9b\x17\xb51`7\xb6\x9b\xb2\xfa]L\x8a\xc3\x1e\xdb3\x94\x04n\xc0k\xbd\xac\xc5}\xa6\xa7V\xc5', 114, 0, b'V\xa5\xbb\x95k|\x8d\x9e\x00\x00')
preset_psk_read(3137470465, 32, 0)
write_firmware(0, b'z\xda\x8c\xdadb\x00\x00\r\x82NV\n\x12\t\x121\x80\n\x002\xc2\xb0\x12\x18K\xb0\x12\xccB\xb0\x12HK\xb0\x12"C\xb0\x12\xa0R=@j\x18<@\x00P\xb0\x12\xe6B=@\x00\x0c<@\x88\x18\xb0\x12(v}@B\x00\xbd\xd0\x10\x00\x00\x00|@D\x00\xbc\xf0\xff\xfd\x00\x00\xbd\xd0 \x00\x00\x00\xbc\xf0\xff\xfb\x00\x00\xb0\x12\xa2V2\xd2:@\xf8I|@\xc0\xff\x8a\x12|@\xc8\xff\x8a\x12:@\xc4\xb49@\xa8\x99\x8a\x12\\B \x04\\\x93\x08 \xc2C \x04~@\n\x00\rA<@l&\x89\x12\\B!\x04\\\x93\x04 \xb0\x12\x98\xc0\xc2C!\x04\\B\x1c\x04\x0c\x93\x1a |@\x05\x00\\\x92\x87\x05\xe2/\\B\x87\x05\xf2\x92\x87\x05_$}BM\x9c)(|\x90\x03\x00Q$}@\x03\x00M\x9c\x18(l\x93G$\xc2C\x8c\x050@t@\x1c\x93\xe9#\xc2\x93\xa3$\x02$\xb0\x12\x84b\xf2\x90\n\x00\x87\x05\x03 \xd2\x93\x88\x05', 2)
write_firmware(256, b'\x1b$\xb0\x12\xe6Y0@\xb0@|\x90\x06\x006$|\x90\x07\x00\xe4#\xb0\x12\xc2Y0@t@|\x90\x0c\x008$}@\x0c\x00M\x9c\x11(|\x90\t\x00-$|\x90\n\x00\xd3#\xb0\x12^O\xd2\x93\x9e\x14\x9f#\xc2C\x9e\x141P\n\x000@2\xf4|\x90\r\x00$$|\x90\x0e\x00\xc2#\xd2\x93\x8c\x05\x90#\xc2C\x8c\x05\xb0\x12\xec\xc60@t@\xb0\x12&Z0@t@\xb0\x12\xf6[0@t@\xb0\x12\x90Y0@t@\xb0\x12\x9cK0@t@\xb0\x12 G0@t@\xb0\x12\xfaQ0@t@\xb0\x12\xc8d0@t@\x0c\x12\xb2\xf0\xff\xfb>\x02|@\x84\x00\xa2L\xbc$\xbc@\x00\x08\x00\x00<A\x00\x13\x0f\x12\x0e\x12\r\x12\x0c\x12\x0b\x12\n\x12\t\x12\x08\x12\x07\x12\x06\x12\x05\x12\x04\x12\xb2@\x00\x04\x84\x00\xb0\x12\x16b\xd2C\x9f\x144A5A6A7A8A9A:A;A<A=A>A?A\x00\x13\x1cB', 2)
write_firmware(512, b'\xa4\x14}@\x13\x00\r\x9c\x04(\x1cS\x82L\xa4\x140A\x82C\xa4\x14\xc2C\xa6\x140@\x04B\x0f\x12\x0e\x12\r\x12\x0c\x12\x0b\x12\n\x12\t\x12\x08\x12\x07\x12\x06\x12\x05\x12\x04\x12:@\x00\x01\xaa\xd3\x00\x00\xa2C\x84\x00\xb0\x12\xf2A\xb0\x12\xbab\x8a\xda\x00\x004A5A6A7A8A9A:A;A<A=A>A?A\x00\x13\xb2@\x07\x00R\x00|@\xc8\x00\xb0\x12\xce\xb4<@&\x01\xb0\x12(U\x82L\x9c\x14\x1dB\x9c\x14<@$\x01\xb0\x12\xf4T\x1cB\x9c\x14\x0c\x93\x02$\xd2C\x9a\x140A\x0f\x12\x0e\x12\r\x12\x0c\x12\x0b\x12\n\x12\t\x12\x08\x12\x07\x12\x06\x12\x05\x12\x04\x12\xb2@\x00\x02\x84\x00\xb0\x12\\B4A5A6A7A8A9A:A;A<A=A>A?A\x00\x13\xb2@\x12B\xc2?\xb2@\x90B\xd2?\xb2@\xb2A\xd4?\xb2@\x98A\xd6?0A\xb2\xd0\x00\x01B\x00>@\x00\x01\xbe\xf0\xff\xfd', 2)
write_firmware(768, b'\x00\x00\xbe\xf0\xff\x8f\x00\x00\x8e\xdc\x00\x00=\xe3\x82M\x02\x01\x9e\xd3\x00\x00\xa2\xd3\x80\x000A<@\x00\x01\xac\xd3\x00\x00\xa2C\x84\x00\x8c\xdc\x00\x000A\n\x12\xb2@\xf3\xfeL\x00\xb2@\xfc\xfdN\x00\xb2\xd0@\x00B\x00\xb2\xf0\xff\xefD\x00<@`\x01\xac\xd2\x00\x00\xb2\xd0\x10\x00l\x01\xb2\xd0\x00\x04\x80\x00\xbc\xd0\x10\x00\x00\x00\xbc\xc2\x00\x00\\BC\x05<\x90S\x00\x1b \xd2C\xa8\x14ML}P\xae\xff<@b\x01^CN\x9d\x15(\xbc\xd2\x00\x00:@`\x01\x9a\xc3\x00\x00\xb0\x12\xaaC\xaa\xd3\x00\x00\xa2\xd3P\x00\xb2\xd0\x00\x02\x80\x00:A0A\xc2C\xa8\x140@hC\xbc\xc2\x00\x000@|C\n\x12:@b\x01\x9a\xc3\x00\x00|@\n\x00\xb0\x12\xd8\xb4\x9a\xd3\x00\x00:A0A\n\x12\t\x12\x08\x12\x07\x12\x06\x12\tL\x08M\x07NFC\x06\x98\x02(0@,\xf4ZG\x03\x00\nZ\nZ\nZ\nZ\nZ\nZ\nZ\nZ\\G', 2)
write_firmware(1024, b"\x02\x00\n\\\x0cF\xb0\x12X\xf5\x0c\\\x0cY_G\x01\x00\x0f_\x0f_\x0f_\x0f_\x0f_\x0f_\x0f_\x0f_mG\r_\x8cM\x00\x00\x8cJ\x02\x00&R'R0@\xd6C\n\x12\t\x12\x08\x12\x07\x12\x06\x12\x05\x12\x04\x121\x80\n\x00\x05L\x0c\x93\x90$^L\x01\x00]L\x02\x00TL\x03\x00VL\x04\x00\\L\x05\x00\x81L\x04\x00YE\x06\x00\\E\x07\x00\x81L\x06\x00XE\x08\x00\\E\t\x00\x81L\x08\x00WE\n\x00\xb2@l\x16\xa0\x15\xb2@\xe6\x15\xa2\x15\xb2@0\x18\xa4\x15\xb2@\xd8\x17\xa6\x15\xb2@H\x15\xa8\x15\xb2@\x10\x15\xac\x15\xb2@\xae\x15\xaa\x15:@>\x16\x0cM\x81M\x00\x00\x81N\x02\x00\xb0\x12X\xf5<\xf0\xff\x00\x8aL\x00\x00\x0cF\xb0\x12X\xf5<\xf0\xff\x00\x8aL\x02\x00\x0cI\xb0\x12X\xf5<\xf0\xff\x00\x8aL\x04\x00\x8aL\x06\x00\x8aL\x08\x00\x0cG\xb0\x12X\xf5<\xf0\xff\x00\x8aL\x0c\x00\x0cH\xb0\x12X\xf5<\xf0", 2)
write_firmware(1280, b'\xff\x00\x8aL\n\x00:@\xc4C\x1eA\x02\x00\x0eU-A<@l\x16\x8a\x12\x0eE\x0eT\rF<@\xe6\x15\x8a\x12\x16A\x04\x00\x06U\x0eF\rI<@0\x18\x8a\x12\x0eF\rI<@\xd8\x17\x8a\x12\x0eF\rI<@H\x15\x8a\x12\x1eA\x08\x00\x0eU\rG<@\x10\x15\x8a\x12\x1eA\x06\x00\x0eU\rH<@\xae\x15\x8a\x12\\C1P\n\x000@(\xf40@ZE\n\x12\t\x12jL\xb0\x12*D<\xf0\xff\x00\x0c\x93I$\x0cJ\xb0\x12L\xf5<\xf0\xff\x00}@\n\x00M\x9cG(\rL\r\\\r]\r]\r\x8c\r]=P\x00\x14\x92M\x02\x00\x88$:@\xfc\x14\x9aM\x04\x00\x08\x00\x9aM\x06\x00\n\x00\x9aM\x08\x00\x0c\x00\xdaM\n\x00\x11\x00\xdaM\x0b\x00\x12\x00\xdaM\x0c\x00\x10\x00\xdaM\r\x00\x13\x00\tJ<\x90\x07\x00\x0c \xb0\x12\xa2V|@X\x00\xb0\x12(U<\xf0\x00\x02\x89L\x06\x000@\x02Fl\xc3l\x92\r ', 2)
[... cut ...]
Firmware: GFUSB_GM168SEC_APP_10019
preset_psk_read(3137470465, 32, 0)
Valid PSK: True
reset(True, False, 20)
read_sensor_register(0, 4)
read_otp()
Traceback (most recent call last):
  File "/home/unjordi/code/goodix-fp-dump/run_521d.py", line 3, in <module>
    driver_52xd.main(0x521d)
  File "/home/unjordi/code/goodix-fp-dump/driver_52xd.py", line 351, in main
    run_driver(device)
  File "/home/unjordi/code/goodix-fp-dump/driver_52xd.py", line 134, in run_driver
    raise ValueError("Invalid OTP")
ValueError: Invalid OTP
`

gsuzeda, 2025-01-14:
Same error, Same device. 
__init__(21021, <class 'protocol.USBProtocol'>, 5)
Found Goodix device: "FingerPrint" from "Goodix" on bus 3 address 55.
Found interface data: 0
Found endpoint in: 0x83
Found endpoint out: 0x1
empty_buffer()
nop()
firmware_version()
Firmware: GFUSB_GM168SEC_APP_10019
preset_psk_read(3137470465, 32, 0)
Valid PSK: True
reset(True, False, 20)
read_sensor_register(0, 4)
read_otp()
Traceback (most recent call last):
  File "/home/gsuzeda/goodix-521d-explanation/goodix-fp-dump/run_521d.py", line 3, in <module>
    driver_52xd.main(0x521d)
    ~~~~~~~~~~~~~~~~^^^^^^^^
  File "/home/gsuzeda/goodix-521d-explanation/goodix-fp-dump/driver_52xd.py", line 351, in main
    run_driver(device)
    ~~~~~~~~~~^^^^^^^^
  File "/home/gsuzeda/goodix-521d-explanation/goodix-fp-dump/driver_52xd.py", line 134, in run_driver
    raise ValueError("Invalid OTP")
ValueError: Invalid OTP

I'm using rog 14 (2021) with fedora 41

egormanga, 2025-01-15:
@unjordi You should not really paste such long logs without proper quoting, see how many people you've accidentally mentioned. You shouldn't have pasted all your reader data either, as this might (somewhat) compromise your security. And, you shouldn't even post that much logs in any case.

Now to the error, it occurs here: https://github.com/goodix-fp-linux-dev/goodix-fp-dump/blob/cc43bb3b3154a0bccc0412ae024013c7e1923139/driver_52xd.py#L133-L134
Try adding a `print(otp)` right above it, so we can better understand what kind of data it gets. Also, be sure not to post your entire OTP string unmodified (unmasked) for the sake of your security.

hjarvi, 2025-01-26:
Hi,

I also have:
```
Bus 005 Device 004: ID 27c6:521d Shenzhen Goodix Technology Co.,Ltd. FingerPrint
```

Value of otp is b''.

What to do to debug this deeper?

```
Firmware: GFUSB_GM168SEC_APP_10019
preset_psk_read(3137470465, 32, 0)
Valid PSK: True
reset(True, False, 20)
read_sensor_register(0, 4)
read_otp()
ERROR
b''
Traceback (most recent call last):
  File "/home/jermu/projects/asus/fingerprint/goodix-fp-dump/run_521d.py", line 3, in <module>
    driver_52xd.main(0x521d)
  File "/home/jermu/projects/asus/fingerprint/goodix-fp-dump/driver_52xd.py", line 353, in main
    run_driver(device)
  File "/home/jermu/projects/asus/fingerprint/goodix-fp-dump/driver_52xd.py", line 136, in run_driver
    raise ValueError("Invalid OTP")
ValueError: Invalid OTP
``

unjordi, 2025-02-15:
> [@unjordi](https://github.com/unjordi) You should not really paste such long logs without proper quoting, see how many people you've accidentally mentioned. You shouldn't have pasted all your reader data either, as this might (somewhat) compromise your security. And, you shouldn't even post that much logs in any case.
> 
> Now to the error, it occurs here:
> 
> [goodix-fp-dump/driver_52xd.py](https://github.com/goodix-fp-linux-dev/goodix-fp-dump/blob/cc43bb3b3154a0bccc0412ae024013c7e1923139/driver_52xd.py#L133-L134)
> 
> Lines 133 to 134 in [cc43bb3](/goodix-fp-linux-dev/goodix-fp-dump/commit/cc43bb3b3154a0bccc0412ae024013c7e1923139)
>  if len(otp) < 64: 
>      raise ValueError("Invalid OTP") 
> 
> Try adding a `print(otp)` right above it, so we can better understand what kind of data it gets. Also, be sure not to post your entire OTP string unmodified (unmasked) for the sake of your security.

Thanks for the feedback! I had no Idea and since then I've given up on the fingerprint reader :/

I'll give it another shot following your suggestion.

barsikus007, 2025-05-13:
same here, suppressing checking (as mentioned in previous issues) leads to random fingerprint scan results

CHAGER5000, 2026-10-02:
I ran into the same empty-OTP failure on 27c6:521d (firmware GFUSB_GM168SEC_APP_10019). Opened a fix here, tested with 5 consecutive real-finger captures: https://github.com/goodix-fp-linux-dev/goodix-fp-dump/pull/81