goodix-fp-linux-dev/goodix-fp-dump #48: ConnectionRefusedError after running without openssl

acovaci, 2024-12-24:
Hi,

I was trying to run this tool on NixOS. However, since OpenSSL was not listed as a dependency, the first time I ran it was without it. This is as far as it got:

```log
update_firmware(0, 57328, 375835607, b"\xea\x0c\x99N\xcb\xfaQ\xaaw-\xfd'\xc0\x9d\xb5v\x8aM\x1d\xa1C\xaf\xbaT\xe2\xe2\xd6\x82\x10\x03\x03\xd1")
reset(False, True, 100)
disconnect()
__init__(21940, <class 'protocol.USBProtocol'>, 5)
Found Goodix device: "Goodix FingerPrint Device" from "Generic" on bus 1 address 7.
Found interface data: 0
Found endpoint in: 0x82
Found endpoint out: 0x1
empty_buffer()
nop()
firmware_version()
Firmware: GF3268_RTSEC_APP_10041
preset_psk_read(3137470471, None, None)
Valid PSK: True
get_iap_version(25)
IAP: MILAN_RTSEC_IAP_10027
Traceback (most recent call last):
  File "/home/acovaci/Code/libfprint-goodixtls-55x4/goodix-fp-dump/run_55b4.py", line 3, in <module>
    driver_55x4.main(0x55b4)
  File "/home/acovaci/Code/libfprint-goodixtls-55x4/goodix-fp-dump/driver_55x4.py", line 258, in main
    run_driver(device)
  File "/home/acovaci/Code/libfprint-goodixtls-55x4/goodix-fp-dump/driver_55x4.py", line 118, in run_driver
    tls_server = subprocess.Popen([
                 ^^^^^^^^^^^^^^^^^^
  File "/nix/store/zv1kaq7f1q20x62kbjv6pfjygw5jmwl6-python3-3.12.7/lib/python3.12/subprocess.py", line 1026, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/nix/store/zv1kaq7f1q20x62kbjv6pfjygw5jmwl6-python3-3.12.7/lib/python3.12/subprocess.py", line 1955, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'openssl'
```

Afterwards, I included OpenSSL as a dependency and tried t rerun it, however now I'm getting a Connection refused error:

```log
Type 2946 to continue and confirm that you are not a bot: 2946
__init__(21940, <class 'protocol.USBProtocol'>, 5)
Found Goodix device: "Goodix FingerPrint Device" from "Generic" on bus 1 address 7.
Found interface data: 0
Found endpoint in: 0x82
Found endpoint out: 0x1
empty_buffer()
nop()
firmware_version()
Firmware: GF3268_RTSEC_APP_10041
preset_psk_read(3137470471, None, None)
Valid PSK: True
get_iap_version(25)
IAP: MILAN_RTSEC_IAP_10027
reset(True, False, 20)
read_sensor_register(0, 4)
read_otp()
Traceback (most recent call last):
  File "/home/acovaci/Code/libfprint-goodixtls-55x4/goodix-fp-dump/run_55b4.py", line 3, in <module>
    driver_55x4.main(0x55b4)
  File "/home/acovaci/Code/libfprint-goodixtls-55x4/goodix-fp-dump/driver_55x4.py", line 258, in main
    run_driver(device)
  File "/home/acovaci/Code/libfprint-goodixtls-55x4/goodix-fp-dump/driver_55x4.py", line 135, in run_driver
    tls_client.connect(("localhost", 4433))
ConnectionRefusedError: [Errno 111] Connection refused
```

I would highly appreciate any ideas on how to approach fixing this. Thanks

acovaci, 2024-12-24:
Also, in case anyone needs it in the future, this is my `shell.nix`, should be an easy `nix-shell` then just `sudo python run_xxxx.py`:

```nix
{
  pkgs ? import <nixpkgs> { },
}:

pkgs.mkShell {
  packages = [
    (pkgs.python312.withPackages (
      ps: with ps; [
        pyusb
        crcmod
        python-periphery
        spidev
        pycryptodome
        crccheck
      ]
    ))
  ];

  nativeBuildInputs = with pkgs; [
    openssl
  ];
}
```

closed 2024-12-24 by acovaci

JohnVeness, 2025-05-18:
> Update: Okay, so turns out we're opening the socket before the openssl server is done initialising. A simple `time.sleep(1)` after the subprocess opening is enough.

Hi, thanks for this. I also had to add the sleep otherwise I was getting the ConnectionRefusedError. @mpi3d Do you want to add the sleep to the code in the repo? Or somehow otherwise deal with the ConnectionRefusedError?

mpi3d, 2025-05-20:
> Hi, thanks for this. I also had to add the sleep otherwise I was getting the ConnectionRefusedError. @mpi3d Do you want to add the sleep to the code in the repo? Or somehow otherwise deal with the [...]

Yeah, why not, if adding some sleep can make it work.