goodix-fp-linux-dev/goodix-fp-dump #49: source .venv/bin/activate

MetalParsley753, 2025-01-13:
when I run the commands on the read me page the command: `source .venv/bin/activate` outputs this:
```
.venv/bin/activate (line 40): “case” builtin not inside of switch block
case "$(uname)" in
^~~^
from sourcing file .venv/bin/activate
source: Error while reading file '.venv/bin/activate'
```
I'm probably just missing something if anyone knows how to fix it.
I am using arch Linux if that makes a difference

egormanga, 2025-01-13:
Probably that's your local environment issue. Try recreating the `.venv` directory (e.g. `rm -r .venv && python -m venv .venv`).

AlexisTM, 2025-01-13:
Sounds to me you've an old version of Python. Update to a newer version which has "case" as a feature.

egormanga, 2025-01-15:
@AlexisTM This is an error from the shell (presumably, `bash`), not Python. Take a look at the syntax: `case "$(uname)" in`

MetalParsley753, 2025-01-18:
I have tried running it in both bash and fish
this is the full out put
`python -m venv .venv
source .venv/bin/activate
pip install -r requirements.txt
sudo lsusb -vd "27c6:" | grep "idProduct" # Returns the device ID
sudo python3 run_5110.py # Change "5110" to your device I`

MetalParsley753, 2025-01-18:
this is the section that seems to be creating the problem and since its for windows I tried commenting it out bu it just keeps un-commenting its self each time I run the command.
here is the section of code
```
# on Windows, a path can contain colons and backslashes and has to be converted:
case "$(uname)" in
    CYGWIN*|MSYS*|MINGW*)
        # transform D:\path\to\venv to /d/path/to/venv on MSYS and MINGW
        # and to /cygdrive/d/path/to/venv on Cygwin
        VIRTUAL_ENV=$(cygpath /home/daniels/goodix-fp-dump/.venv)
        export VIRTUAL_ENV
        ;;
    *)
        # use the path as-is
        export VIRTUAL_ENV=/home/daniels/goodix-fp-dump/.venv
        ;;
esac
```

MetalParsley753, 2025-01-19:
it was a python issue i just had to use the fish script

closed 2025-01-19 by MetalParsley753