goodix-fp-linux-dev/goodix-fp-dump #53: "IndexError: index out of range" for image chunk[5]

JohnVeness, 2025-05-18:
Hi, I'm trying to get "run_55b4.py" working on my Lenovo IdeaPad S540-15IWL GTX, in openSUSE Tumbleweed.

I've got this error:
```
Type 1523 to continue and confirm that you are not a bot: 1523
__init__(21940, <class 'protocol.USBProtocol'>, 5)
Found Goodix device: "Goodix FingerPrint Device" from "Generic" on bus 1 address 6.
Found interface data: 0
Found endpoint in: 0x82
Found endpoint out: 0x1
empty_buffer()
nop()
firmware_version()
Firmware: GF3268_RTSEC_APP_10041
preset_psk_read(3137470471, None, None)
Valid PSK: True
get_iap_version(25)
IAP: MILAN_RTSEC_IAP_10027
reset(True, False, 20)
read_sensor_register(0, 4)
read_otp()
request_tls_connection()
upload_config_mcu(b'`\x11`q$\x95,\xc1\x14\xd5\x10\xe5\x00\xe5\x14\xf9\x03\x04\x02\x00\x00\x08\x00\x11\x11\xba\x00\x01\x80\xca\x00\x07\x00\x84\x00\xc0\xb3\x86\x00\xbb\xc4\x88\x00\xba\xba\x8a\x00\xb2\xb2\x8c\x00\xaa\xaa\x8e\x00\xc1\xc1\x90\x00\xbb\xbb\x92\x00\xb1\xb1\x94\x00\x00\xa8\x96\x00\x00\xb6\x98\x00\x00\xbf\x9a\x00\x00\xbaP\x00\x01\x05\xd0\x00\x00\x00p\x00\x00\x00r\x00xVt\x004\x12&\x00\x00\x12 \x00\x10@\x12\x00\x03\x04*\x01\x02\x00"\x00\x01 $\x002\x00\x80\x00\x01\x00\\\x00\x80\x00V\x00\x08 X\x00\x01\x002\x00,\x02\x82\x00\x80\x0c\xba\x00\x01\x80\xca\x00\x07\x00*\x01\x82\x03 \x00\x10@"\x00\x01 $\x00\x14\x00\x80\x00\x05\x00\\\x00\x00\x01V\x00\x08 X\x00\x03\x00\x82\x00\x80\x14*\x01\x08\x00\\\x00\x80\x00b\x00\t\x03d\x00\x18\x00"\x00\x00 *\x01\x08\x00\\\x00\x00\x01R\x00\x08\x00T\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x9ai')
mcu_switch_to_fdt_mode(b'\r\x01\x80\x12\x80\x12\x80\x98\x80\x82\x80\x12\x80\xa0\x80\x99\x80\x7f\x80\x12\x80\x9f\x80\x93\x80~', True)
mcu_get_image()
Traceback (most recent call last):
  File "/home/john/coding/goodix-fp-dump/run_55b4.py", line 3, in <module>
    driver_55x4.main(0x55b4)
    ~~~~~~~~~~~~~~~~^^^^^^^^
  File "/home/john/coding/goodix-fp-dump/driver_55x4.py", line 261, in main
    run_driver(device)
    ~~~~~~~~~~^^^^^^^^
  File "/home/john/coding/goodix-fp-dump/driver_55x4.py", line 158, in run_driver
    tool.decode_image(tls_server.stdout.read(14260)[:-4]),
    ~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/john/coding/goodix-fp-dump/tool.py", line 43, in decode_image
    image.append(((chunk[5] & 0xf) << 8) + chunk[2])
                   ~~~~~^^^
IndexError: index out of range
```

Any ideas? Many thanks for your work on this :)

TheWeirdDev, 2025-05-18:
Can you add print statements to the `decode_image` function and see what `len(data)` is?

JohnVeness, 2025-05-18:
Thanks for quick reply.

With:
```
def decode_image(data: bytes):
    image: list[int] = []
    print("len(data)=", len(data))
    for i in range(0, len(data), 6):
        chunk = data[i:i + 6]

        image.append(((chunk[0] & 0xf) << 8) + chunk[1])
        image.append((chunk[3] << 4) + (chunk[0] >> 4))
        image.append(((chunk[5] & 0xf) << 8) + chunk[2])
        image.append((chunk[4] << 4) + (chunk[5] >> 4))

    return image
```

I get output:

```
Type 4849 to continue and confirm that you are not a bot: 4849
__init__(21940, <class 'protocol.USBProtocol'>, 5)
Found Goodix device: "Goodix FingerPrint Device" from "Generic" on bus 1 address 6.
Found interface data: 0
Found endpoint in: 0x82
Found endpoint out: 0x1
empty_buffer()
nop()
firmware_version()
Firmware: GF3268_RTSEC_APP_10041
preset_psk_read(3137470471, None, None)
Valid PSK: True
get_iap_version(25)
IAP: MILAN_RTSEC_IAP_10027
reset(True, False, 20)
read_sensor_register(0, 4)
read_otp()
request_tls_connection()
upload_config_mcu(b'`\x11`q$\x95,\xc1\x14\xd5\x10\xe5\x00\xe5\x14\xf9\x03\x04\x02\x00\x00\x08\x00\x11\x11\xba\x00\x01\x80\xca\x00\x07\x00\x84\x00\xc0\xb3\x86\x00\xbb\xc4\x88\x00\xba\xba\x8a\x00\xb2\xb2\x8c\x00\xaa\xaa\x8e\x00\xc1\xc1\x90\x00\xbb\xbb\x92\x00\xb1\xb1\x94\x00\x00\xa8\x96\x00\x00\xb6\x98\x00\x00\xbf\x9a\x00\x00\xbaP\x00\x01\x05\xd0\x00\x00\x00p\x00\x00\x00r\x00xVt\x004\x12&\x00\x00\x12 \x00\x10@\x12\x00\x03\x04*\x01\x02\x00"\x00\x01 $\x002\x00\x80\x00\x01\x00\\\x00\x80\x00V\x00\x08 X\x00\x01\x002\x00,\x02\x82\x00\x80\x0c\xba\x00\x01\x80\xca\x00\x07\x00*\x01\x82\x03 \x00\x10@"\x00\x01 $\x00\x14\x00\x80\x00\x05\x00\\\x00\x00\x01V\x00\x08 X\x00\x03\x00\x82\x00\x80\x14*\x01\x08\x00\\\x00\x80\x00b\x00\t\x03d\x00\x18\x00"\x00\x00 *\x01\x08\x00\\\x00\x00\x01R\x00\x08\x00T\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x9ai')
mcu_switch_to_fdt_mode(b'\r\x01\x80\x12\x80\x12\x80\x98\x80\x82\x80\x12\x80\xa0\x80\x99\x80\x7f\x80\x12\x80\x9f\x80\x93\x80~', True)
mcu_get_image()
len(data)= 221
Traceback (most recent call last):
  File "/home/john/coding/goodix-fp-dump/run_55b4.py", line 3, in <module>
    driver_55x4.main(0x55b4)
    ~~~~~~~~~~~~~~~~^^^^^^^^
  File "/home/john/coding/goodix-fp-dump/driver_55x4.py", line 261, in main
    run_driver(device)
    ~~~~~~~~~~^^^^^^^^
  File "/home/john/coding/goodix-fp-dump/driver_55x4.py", line 158, in run_driver
    tool.decode_image(tls_server.stdout.read(14260)[:-4]),
    ~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/john/coding/goodix-fp-dump/tool.py", line 44, in decode_image
    image.append(((chunk[5] & 0xf) << 8) + chunk[2])
                   ~~~~~^^^
IndexError: index out of range
```

That is, len(data)=221.

JohnVeness, 2025-05-18:
Note that I had to add `time.sleep(1)` in `run_driver` as follows:

```
def run_driver(device: goodix.Device):
    tls_server = subprocess.Popen([
        "openssl", "s_server", "-nocert", "-psk",
        PSK.hex(), "-port", "4433", "-quiet"
    ],
                                  stdout=subprocess.PIPE,
                                  stderr=subprocess.STDOUT)

    time.sleep(1)

    try:
        if not device.reset(True, False, 20)[0]:
            raise ValueError("Reset failed")
...
```
(and `import time` at the top)

in order to avoid a ConnectionRefusedError. See #48

TheWeirdDev, 2025-05-18:
That's very strange. The data should be a lot bigger than that.

Add another `time.sleep(1)` before `tool.write_pgm` on line 154

If that didn't work try this script: [goodix_55b4_nofw_0psk.py](https://gist.github.com/TheWeirdDev/b702d5c90c87c88d1ab54d1de80d7d56) (you need opencv and matplotlib libraries for python)

JohnVeness, 2025-05-18:
The extra time.sleep(1) made no difference. The other script outputs:
```
device configs:
(<CONFIGURATION 1: 100 mA>,)
We only have one, print its interfaces
(<INTERFACE 0: Vendor Specific>,)
Again, only one. Lets view its endpoints
    INTERFACE 0: Vendor Specific ===========================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x0
     bAlternateSetting  :    0x0
     bNumEndpoints      :    0x2
     bInterfaceClass    :   0xff Vendor Specific
     bInterfaceSubClass :    0x0
     bInterfaceProtocol :    0x0
     iInterface         :    0x0 
      ENDPOINT 0x1: Bulk OUT ===============================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :    0x1 OUT
       bmAttributes     :    0x2 Bulk
       wMaxPacketSize   :  0x200 (512 bytes)
       bInterval        :    0x0
      ENDPOINT 0x82: Bulk IN ===============================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x82 IN
       bmAttributes     :    0x2 Bulk
       wMaxPacketSize   :  0x200 (512 bytes)
       bInterval        :    0x0
Reset
read reg
send nop
read otp
mcu download chip config
mcuSwitchToFdtMode
send nop
Getting FW Version
Found expected firmware! b'GF3268_RTSEC_APP_10041\x00'
PresetPskReadR
Chip already uses our PSK!
request TLS connect. FP will send client hello back.
160303002f0100002b0303c4a1b38262f2f668119d7670ed7bbdabab6aa9bf987428b64a86ec4bad5b37c300000400ae00ff0100
connecting...
got server_hello + server_hello_done as  16030300510200004d0303f8a72bfbda0be69073c6201d989c5d16bc0d7862ff81302e444f574e47524401203f725b0c27075a2cc469f988725933edc701da48ffa0646bb75327dec21cfc2800ae000005ff0100010016030300040e000000
Device initialization and TLS connection complete!
McuGetImage
Image:  b'40A7D09F617F0000:error:80000062:system library:BIO_bind:Address already in use:crypto/bio/bio_sock2.c:240:calling bind()\n40A7D09F617F0000:error:10000075:BIO routines:BIO_bind:unable to bind socket:crypto/bio/bio_sock2.c:242:\n' 343041374430394636313746303030303a6572726f723a38303030303036323a73797374656d206c6962726172793a42494f5f62696e643a4164647265737320616c726561647920696e207573653a63727970746f2f62696f2f62696f5f736f636b322e633a3234303a63616c6c696e672062696e6428290a343041374430394636313746303030303a6572726f723a31303030303037353a42494f20726f7574696e65733a42494f5f62696e643a756e61626c6520746f2062696e6420736f636b65743a63727970746f2f62696f2f62696f5f736f636b322e633a3234323a0a
```

JohnVeness, 2025-05-18:
Ah, hang on, the "Address already in use" gave me a clue. It seems openssl was still running after previous attempts. Killing the process made run_55b4.py now work, and create a fingerprint.pgm file. I had to change time.sleep(0.1) to time.sleep(1) in goodix_55b4_nofw_0psk.py to get that to work, but it now does. Although I'm not sure what it does - it seems to read 1000 images but I can't see any new output files..? Thanks anyway!